SolderedElectronics/Inkplate-Arduino-library #268: NetworkClient::downloadFileHTTPS ignores certificate validility

cyruscook, 2025-03-05:
The API method `NetworkClient::downloadFileHTTPS` ignores the validity of the server's HTTP certificate, [by calling](https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/2dd263f2f9548aadac8638413a143beddf068a64/src/include/NetworkClient.cpp#L400) `NetworkClientSecure::setInsecure` which ultimately passes the `MBEDTLS_SSL_VERIFY_NONE` to the Mbed TLS library. The effect of this is that the servers certificate is not checked whatsoever, it could be garbage data, it could be for a different domain, or it could have expired, and the method will still trust the server.

This behaviour is not documented at all, so users of the library will probably assume, wrongly, that the `downloadFileHTTPS` method will securely check the HTTPS certificate of the server.

The [documentation for the Mbed TLS library](https://mbed-tls.readthedocs.io/en/latest/kb/how-to/mbedtls-tutorial/#configuring-ssl-tls) says "Do not use this in a full application".

It is also worth noting that Mbed TLS are planning to disable this behaviour in version 4.0.0: https://github.com/Mbed-TLS/mbedtls/issues/7080

It would be a breaking change if the `setInsecure` call was removed, so perhaps you will not be keen to do so, but it would be advisable to at least update the documentation to make very clear the security risks of this API call, and list potential alternatives for secure HTTPS calls (e.g. directly using WiFiClientSecure).

rsoric, 2025-03-13:
Hi @cyruscook,

you're absolutely correct, we will look into also adding an override of this function where you can supply a https certificate and also update it in our docs. We are in the process of updating all Inkplate docs, so far only for Inkplate 6 MOTION, so it will be added in the [new docs](https://docs.inkplate.com/). Unsure when this will be exactly, so I'll leave the issue open until then.

Thank you for pointing this out to us to help make Inkplate better! 

-Rob

rsoric, 2025-05-16:
Hi @cyruscook ,

we have pushed a new release of the Inkplate library where you can use a [HTTPS certificate](https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/master/examples/Inkplate10/Advanced/WEB_WiFi/Inkplate10_HTTPS_With_Certificate/Inkplate10_HTTPS_With_Certificate.ino). So, this was added in the library in the meantime! Hope it helps :)

-Rob

closed 2025-05-16 by rsoric