SolderedElectronics/Inkplate-Arduino-library #270: Rust support?

tirithen, 2025-04-03:
Hi! Are there any plans for, or known projects that uses Rust for controlling the screen of Inkplate 10 devices?

I have had great success connecting to WiFi, making HTTP calls with the [esp-idf-svc ](https://github.com/esp-rs/esp-idf-svc) crate. I have also successfully communicated with the two i/o expander chips. However I have not been able to write code to successfully drive the e-paper display.

There are several promising rust graphics crates to use to render shapes/fonts such as [embedded-graphics](https://crates.io/crates/embedded-graphics), but to use them a basic driver with the correct timings would be needed. I think several other people would find this useful as well.

Are there any plans or known ongoing projects to make this work?

rsoric, 2025-05-16:
Hi @tirithen ,

thanks for your inquiry and really sorry for the delayed reply! It's a good idea but unfortunately we aren't planning to offer rust support for Inkplate anytime soon. Our hardware and software (low level e-paper drivers in the Inkplate library) is always totally open so you are free to attempt to make a contribution :)

Best
-Rob

closed 2025-05-16 by rsoric

tirithen, 2025-05-16:
@rsoric Thanks for taking the time to answer! I can see how it is tricky to find the time. I might give it a a go later on, but I got stuck in my previous attempts so might be some time.

If anyone else on the internet would be interesting with more experience in embed rust, I'll definitely be up for testing, and also improving if I can get help get the most basic driver working. ;-)