SolderedElectronics/Inkplate-Arduino-library #271: Manual reset not working as expected

tatulea, 2025-05-11:
Hello, I am putting the device in deep sleep and I expect the I/O button (next to the USB connection) to turn off/on (reset) the ESP. What actually happens is that I push the button once, the LED goes off, I push it back, but ESP does not start until the deep sleep gets triggered (used `esp_sleep_enable_timer_wakeup`).

I would expect the power button to reset the ESP.

rsoric, 2025-05-16:
Hi @tatulea ,

if I understand correctly,

- Inkplate is powered via battery and put into deep sleep via software
- You are resetting via the power button while Inkplate is in deep sleep
- The device is not waking up after two presses of the power button (power off - power on)

You are right to expect that in this case Inkplate should do a software reset.

Does this behavior happen on USB power as well or just battery? Which Inkplate model do you have? I don't believe I've encountered this issue but we can try to replicate it.

-Rob

tatulea, 2025-08-15:
Hey, sorry for the very late reply!

I am using COLOR6. And I am testing with the Inkplate6COLOR_Simple_Deep_Sleep example.

I was also trying to add

```cpp
esp_sleep_enable_ext0_wakeup(GPIO_NUM_39, 0);
```

but it does not seem to work. I can also use `GPIO_NUM_36` to wake it up from the WAKE button.

tatulea, 2025-09-02:
Thanks, that is true. Then what I am not sure about is the turn on/off button. I would have expected that button to turn on and off the entire power supply for the board, forcing a reboot of the ESP, no matter if it was in deep sleep or not. But looks like if it is in deep sleep, it will not reboot. Is that right?

rsoric, 2025-09-04:
Hi @tatulea , can you please take a detailed photo of the back of your board, mainly we are interested in looking at the circuitry around the power switch.