SolderedElectronics/Inkplate-Arduino-library #273: 10.1.0 and 10.2.0 point to same commit

zBart, 2025-05-19:
I think something went wrong with the tagging of the new vesion:

- https://github.com/SolderedElectronics/Inkplate-Arduino-library/tags
- https://github.com/SolderedElectronics/Inkplate-Arduino-library/releases/tag/10.1.0
- https://github.com/SolderedElectronics/Inkplate-Arduino-library/releases/tag/10.2.0

Both tags point at the same commit: 
- https://github.com/SolderedElectronics/Inkplate-Arduino-library/commit/2dd263f2f9548aadac8638413a143beddf068a64

rsoric, 2025-05-20:
Hi @zBart 

thanks for letting us know about this, we'll get it sorted ASAP

rsoric, 2025-05-20:
I've just done a re-release, it's my mistake, I was not aware that the tag also was tied to the commit hash so now it should be OK

Thanks so much for pointing it out to us!¨

-Rob

closed 2025-05-20 by rsoric

rsoric, 2025-05-20:
Thanks again, this worked well and made the update visible in Arduino 

[image]