SolderedElectronics/Inkplate-Arduino-library #274: Inkplate 10 esphome 3 bit greyscale not working as it should.

tomtom13, 2025-05-30:
Hi gents, I know that this might be a stretch, however I'm looking for some support / pointers on displaying 3 bit images through esphome device builder.

ATM, the defined "inkplate6" model "Inkplate_10" only supports binary (black&white) input of images OR full 24bit ... and the 24 bit input doesn't seem to do the right conversion into 3 bit grayscale. 

I'll attach my yaml for esphome, however if somebody could point me to where the code for the display definition I could start drilling into the problem my self.

```
esphome:
  name: inkplate-10-1
  friendly_name: inkplate_10-1

esp32:
  board: esp-wrover-kit

psram:
  mode: quad
  speed: 80MHz

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "REDACTED"

ota:
  - platform: esphome
    password: "REDACTED"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

i2c:
      
time:
  - platform: pcf85063
    id: esptime
    # repeated synchronization is not necessary unless the external RTC
    # is much more accurate than the internal clock
    update_interval: never
  - platform: homeassistant
    # instead try to synchronize via network repeatedly ...
    on_time_sync:
      then:
        # ... and update the RTC when the synchronization was successful
        pcf85063.write_time:

pca6416a:
  - id: pca6416a_hub
    address: 0x20
    # Primary expander for display control and additional I/O
  - id: pca6416a_hub2
    address: 0x21
    # Secondary expander for additional I/O

switch:
  - platform: gpio
    id: battery_read_mosfet
    pin:
      pca6416a: pca6416a_hub
      number: 9

sensor:
  - platform: adc
    id: battery_voltage
    update_interval: never
    attenuation: 12db
    pin: 35
  - platform: template
    name: "Inkplate Battery Voltage"
    unit_of_measurement: "V"
    accuracy_decimals: 3
    lambda: |-
      // Enable MOSFET to connect battery voltage divider
      id(battery_read_mosfet).turn_on();
      // Wait for voltage to stabilize
      delay(5);
      // Sample ADC value
      float adc = id(battery_voltage).sample();
      // Disable MOSFET to save power
      id(battery_read_mosfet).turn_off();
      return adc;
    filters:
      - multiply: 2 # Compensate for voltage divider (1:2 ratio)

http_request:
  id: fetch_image_request
  verify_ssl: false
  useragent: esphome/seeed-studio-display-1
  timeout: 60s
  watchdog_timeout: 3min

online_image:
  - id: dashboard_image
    format: PNG
    type: BINARY
    url: http://REDACTED/dashboard-displays/inkplate10_1?viewport=1200x825&eink=2
    update_interval: 20min
    on_download_finished:
      - delay: 0ms
      - component.update: main_display

display:
  - platform: inkplate6
    id: main_display
    greyscale: false
    partial_updating: false
    update_interval: never
    model: inkplate_10
    lambda: |-
      it.image(0, 0, id(dashboard_image));

    ckv_pin: 32
    sph_pin: 33
    gmod_pin:
      pca6416a: pca6416a_hub
      number: 1
    gpio0_enable_pin:
      pca6416a: pca6416a_hub
      number: 8
    oe_pin:
      pca6416a: pca6416a_hub
      number: 0
    spv_pin:
      pca6416a: pca6416a_hub
      number: 2
    powerup_pin:
      pca6416a: pca6416a_hub
      number: 4
    wakeup_pin:
      pca6416a: pca6416a_hub
      number: 3
    vcom_pin:
      pca6416a: pca6416a_hub
      number: 5

```

rsoric, 2025-06-03:
Hi @tomtom13 ,

as we haven't written the implementations for ESPHome (they are user contributions) I can't help with this issue. We are looking to make official implementations in the near future, and this Arduino library actually only served as a starting base to make the ESPHome implementations and is not really found 'under the hood' in the ESPHome Inkplate devices, so we can't debug it.

It sounds like an error in the implementation end, so see with the author.

Best
-Rob

tomtom13, 2025-06-08:
Wait, if it's a user contribution for ESPhome, how do I trace the author. I'm very fresh to the esp32 and ESPhome so pointers on how the project is structured would be appreciated.